ros2/kilted_tutorial_party #177: Executables in `examples_rclcpp_minimal_subscriber`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for arm64 and amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde