ros2/kilted_tutorial_party #178: Executables in `examples_rclcpp_minimal_subscriber`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine**

Check `subscriber_lambda`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_subscriber subscriber_lambda
[INFO] [1746228532.015142100] [minimal_subscriber]: I heard: 'Hello, world! 0'
[INFO] [1746228532.498339800] [minimal_subscriber]: I heard: 'Hello, world! 1'
[INFO] [1746228532.997669200] [minimal_subscriber]: I heard: 'Hello, world! 2'
[INFO] [1746228533.497810700] [minimal_subscriber]: I heard: 'Hello, world! 3'
[INFO] [1746228534.014181200] [minimal_subscriber]: I heard: 'Hello, world! 4'
[INFO] [1746228534.498943500] [minimal_subscriber]: I heard: 'Hello, world! 5'
[INFO] [1746228534.997763300] [minimal_subscriber]: I heard: 'Hello, world! 6'
[INFO] [1746228535.498449700] [minimal_subscriber]: I heard: 'Hello, world! 7'
[INFO] [1746228536.013577600] [minimal_subscriber]: I heard: 'Hello, world! 8'
[INFO] [1746228536.513341700] [minimal_subscriber]: I heard: 'Hello, world! 9'
[INFO] [1746228536.997679300] [minimal_subscriber]: I heard: 'Hello, world! 10'
[INFO] [1746228537.497859200] [minimal_subscriber]: I heard: 'Hello, world! 11'
```

Check `subscriber_member_function`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_subscriber subscriber_member_function
[INFO] [1746228586.516778200] [minimal_subscriber]: I heard: 'Hello, world! 109'
[INFO] [1746228587.004303600] [minimal_subscriber]: I heard: 'Hello, world! 110'
[INFO] [1746228587.500696400] [minimal_subscriber]: I heard: 'Hello, world! 111'
[INFO] [1746228587.999882200] [minimal_subscriber]: I heard: 'Hello, world! 112'
[INFO] [1746228588.499461700] [minimal_subscriber]: I heard: 'Hello, world! 113'
[INFO] [1746228589.000373600] [minimal_subscriber]: I heard: 'Hello, world! 114'
[INFO] [1746228589.499593000] [minimal_subscriber]: I heard: 'Hello, world! 115'
[INFO] [1746228591.260649600] [minimal_subscriber]: I heard: 'Hello, world! 116'
[INFO] [1746228591.264128300] [minimal_subscriber]: I heard: 'Hello, world! 117'
[INFO] [1746228591.266269900] [minimal_subscriber]: I heard: 'Hello, world! 118'
[INFO] [1746228591.500590200] [minimal_subscriber]: I heard: 'Hello, world! 119'
[INFO] [1746228592.000068400] [minimal_subscriber]: I heard: 'Hello, world! 120'
[INFO] [1746228592.499709600] [minimal_subscriber]: I heard: 'Hello, world! 121'
[INFO] [1746228592.999979800] [minimal_subscriber]: I heard: 'Hello, world! 122'
```

Check `subscriber_not_composable`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable
[INFO] [1746228629.014448200] [minimal_subscriber]: I heard: 'Hello, world! 194'
[INFO] [1746228629.510523900] [minimal_subscriber]: I heard: 'Hello, world! 195'
[INFO] [1746228630.010873900] [minimal_subscriber]: I heard: 'Hello, world! 196'
[INFO] [1746228630.510649500] [minimal_subscriber]: I heard: 'Hello, world! 197'
[INFO] [1746228631.011625400] [minimal_subscriber]: I heard: 'Hello, world! 198'
[INFO] [1746228631.510808200] [minimal_subscriber]: I heard: 'Hello, world! 199'
[INFO] [1746228632.010581700] [minimal_subscriber]: I heard: 'Hello, world! 200'
[INFO] [1746228632.510435000] [minimal_subscriber]: I heard: 'Hello, world! 201'
[INFO] [1746228633.010679500] [minimal_subscriber]: I heard: 'Hello, world! 202'
[INFO] [1746228634.821469200] [minimal_subscriber]: I heard: 'Hello, world! 203'
[INFO] [1746228634.823527200] [minimal_subscriber]: I heard: 'Hello, world! 204'
[INFO] [1746228634.825791000] [minimal_subscriber]: I heard: 'Hello, world! 205'
[INFO] [1746228635.011177600] [minimal_subscriber]: I heard: 'Hello, world! 206'
[INFO] [1746228635.510305800] [minimal_subscriber]: I heard: 'Hello, world! 207'
[INFO] [1746228636.010412700] [minimal_subscriber]: I heard: 'Hello, world! 208'
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde