ros2/kilted_tutorial_party #181: Executables in `examples_rclcpp_multithreaded_executor`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `multithreaded_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed the check for Issue #181 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor`.
- The node started successfully and ran without errors.
- The console output showed interleaved messages from different components/callbacks, including distinct thread IDs in the subscriber messages (e.g., `[DualThreadedNode]: THREAD ... => Heard ...`), confirming that the MultiThreadedExecutor was processing callbacks concurrently.

The executable ran as expected.

ahcorde, 2025-05-05:
Thank you for testing