ros2/kilted_tutorial_party #183: Executables in `examples_rclcpp_multithreaded_executor`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `multithreaded_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine**

```
C:\pixi_ws>ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor
[INFO] [1746228421.845189700] [PublisherNode]:
<<THREAD 3300291727805597874>> Publishing 'Hello World! 0'
[INFO] [1746228421.845917500] [DualThreadedNode]: THREAD 6591353612496544475 => Heard 'Hello World! 0' at 1746228421845910800
[INFO] [1746228421.845916900] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 0' at 1746228421845910800
[INFO] [1746228422.345608300] [PublisherNode]:
<<THREAD 6494238244167718650>> Publishing 'Hello World! 1'
[INFO] [1746228422.347526500] [DualThreadedNode]: THREAD 3300291727805597874 => Heard 'Hello World! 1' at 1746228422347524200
[INFO] [1746228422.347548500] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 1' at 1746228422347547000
[INFO] [1746228422.845074700] [PublisherNode]:
<<THREAD 6591353612496544475>> Publishing 'Hello World! 2'
[INFO] [1746228422.847429900] [DualThreadedNode]: THREAD 6494238244167718650 => Heard 'Hello World! 2' at 1746228422847427400
[INFO] [1746228422.847450400] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 2' at 1746228422847448500
[INFO] [1746228423.345037300] [PublisherNode]:
<<THREAD 3300291727805597874>> Publishing 'Hello World! 3'
[INFO] [1746228423.347344400] [DualThreadedNode]: THREAD 6591353612496544475 => Heard 'Hello World! 3' at 1746228423347341400
[INFO] [1746228423.347344300] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 3' at 1746228423347342300
[INFO] [1746228423.844632700] [PublisherNode]:
<<THREAD 6494238244167718650>> Publishing 'Hello World! 4'
[INFO] [1746228423.844923700] [DualThreadedNode]: THREAD 3300291727805597874 => Heard 'Hello World! 4' at 1746228423844921900
[INFO] [1746228423.844937400] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 4' at 1746228423844935800
[INFO] [1746228424.345211800] [PublisherNode]:
<<THREAD 6591353612496544475>> Publishing 'Hello World! 5'
[INFO] [1746228424.347514500] [DualThreadedNode]: THREAD 6494238244167718650 => Heard 'Hello World! 5' at 1746228424347512300
[INFO] [1746228424.347559200] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 5' at 1746228424347557300
[INFO] [1746228424.845355500] [PublisherNode]:
<<THREAD 3300291727805597874>> Publishing 'Hello World! 6'
[INFO] [1746228424.847491000] [DualThreadedNode]: THREAD 6591353612496544475 => Heard 'Hello World! 6' at 1746228424847487500
[INFO] [1746228424.847497200] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 6' at 1746228424847495700
[INFO] [1746228425.345464100] [PublisherNode]:
<<THREAD 6494238244167718650>> Publishing 'Hello World! 7'
[INFO] [1746228425.347858100] [DualThreadedNode]: THREAD 3300291727805597874 => Heard 'Hello World! 7' at 1746228425347842000
[INFO] [1746228425.347917600] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 7' at 1746228425347915500
[INFO] [1746228425.845282000] [PublisherNode]:
<<THREAD 6591353612496544475>> Publishing 'Hello World! 8'
[INFO] [1746228425.848107000] [DualThreadedNode]: THREAD 6494238244167718650 => Heard 'Hello World! 8' at 1746228425848103600
[INFO] [1746228425.848116600] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 8' at 1746228425848115100
[INFO] [1746228426.344858900] [PublisherNode]:
<<THREAD 3300291727805597874>> Publishing 'Hello World! 9'
[INFO] [1746228426.347257900] [DualThreadedNode]: THREAD 6591353612496544475 => Heard 'Hello World! 9' at 1746228426347252400
[INFO] [1746228426.347267200] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 9' at 1746228426347264400
[INFO] [1746228426.845398000] [PublisherNode]:
<<THREAD 6494238244167718650>> Publishing 'Hello World! 10'
[INFO] [1746228426.846399200] [DualThreadedNode]: THREAD 3300291727805597874 => Heard 'Hello World! 10' at 1746228426846395700
[INFO] [1746228426.846427100] [DualThreadedNode]: THREAD 10742284550632295126 => Heard 'Hello World! 10' at 1746228426846425100
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde