ros2/kilted_tutorial_party #186: Executables in `examples_rclpy_executors`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors callback_group
       ```
  
    
  </details>
- [ ] **Check `composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors composed
       ```
  
    
  </details>
- [ ] **Check `custom_callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_callback_group
       ```
  
    
  </details>
- [ ] **Check `custom_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_executor
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors listener
       ```
  
    
  </details>
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Successfully completed all checks for Issue #186 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run examples_rclpy_executors callback_group`. Node started successfully and ran without errors, showing expected callback group activity.
- Ran `ros2 run examples_rclpy_executors composed`. Node started successfully and ran without errors, showing output indicating composed nodes functioning.
- Ran `ros2 run examples_rclpy_executors custom_callback_group`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_executors custom_executor`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_executors talker` and `ros2 run examples_rclpy_executors listener` simultaneously. Nodes started successfully, and the listener correctly received messages ("I heard...") from the talker.

All Python executor examples ran as expected.