ros2/kilted_tutorial_party #187: Executables in `examples_rclpy_executors`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors callback_group
       ```
  
    
  </details>
- [ ] **Check `composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors composed
       ```
  
    
  </details>
- [ ] **Check `custom_callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_callback_group
       ```
  
    
  </details>
- [ ] **Check `custom_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_executor
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors listener
       ```
  
    
  </details>
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
It works for me

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors callback_group
[INFO] [1746284591.206728022] [double_talker]: Publishing: "Hello World: 0"
[INFO] [1746284591.207722633] [listener]: I heard: "Hello World: 0"
[INFO] [1746284591.689829017] [double_talker]: Publishing: "Hello World: 1"
[INFO] [1746284591.690927565] [double_talker]: Publishing: "Hello World: 2"
[INFO] [1746284591.691035941] [listener]: I heard: "Hello World: 1"
[INFO] [1746284591.691613784] [listener]: I heard: "Hello World: 2"
[INFO] [1746284592.189494468] [double_talker]: Publishing: "Hello World: 3"
[INFO] [1746284592.190431518] [listener]: I heard: "Hello World: 3"
[INFO] [1746284592.689694840] [double_talker]: Publishing: "Hello World: 4"
[INFO] [1746284592.691193234] [double_talker]: Publishing: "Hello World: 5"
[INFO] [1746284592.691383710] [listener]: I heard: "Hello World: 4"
[INFO] [1746284592.692021118] [listener]: I heard: "Hello World: 5"
[INFO] [1746284593.189594978] [double_talker]: Publishing: "Hello World: 6"
[INFO] [1746284593.190612085] [listener]: I heard: "Hello World: 6"
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors composed
[INFO] [1746284598.318884309] [talker]: Publishing: "Hello World: 0"
[INFO] [1746284598.319466423] [listener]: I heard: "Hello World: 0"
[INFO] [1746284599.310990115] [talker]: Publishing: "Hello World: 1"
[INFO] [1746284599.311461696] [listener]: I heard: "Hello World: 1"
[INFO] [1746284600.311025563] [talker]: Publishing: "Hello World: 2"
[INFO] [1746284600.311518773] [listener]: I heard: "Hello World: 2"
[INFO] [1746284601.311013545] [talker]: Publishing: "Hello World: 3"
[INFO] [1746284601.311514815] [listener]: I heard: "Hello World: 3"
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors custom_callback_group
[INFO] [1746284605.139709788] [intermittent_talker]: Publishing: "Hello World: 0"
[INFO] [1746284605.231863020] [intermittent_talker]: Publishing: "Hello World: 1"
[INFO] [1746284605.331879935] [intermittent_talker]: Publishing: "Hello World: 2"
[INFO] [1746284605.431734644] [intermittent_talker]: Publishing: "Hello World: 3"
[INFO] [1746284605.531855691] [intermittent_talker]: Publishing: "Hello World: 4"
[INFO] [1746284605.631825289] [intermittent_talker]: Publishing: "Hello World: 5"
[INFO] [1746284605.731827854] [intermittent_talker]: Publishing: "Hello World: 6"
[INFO] [1746284605.831803240] [intermittent_talker]: Publishing: "Hello World: 7"
[INFO] [1746284605.931927968] [intermittent_talker]: Publishing: "Hello World: 8"
[INFO] [1746284606.031856144] [intermittent_talker]: Publishing: "Hello World: 9"
[INFO] [1746284606.131781526] [intermittent_talker]: Publishing: "Hello World: 10"
[INFO] [1746284606.231780922] [intermittent_talker]: Publishing: "Hello World: 11"
[INFO] [1746284606.531866721] [intermittent_talker]: Publishing: "Hello World: 12"
[INFO] [1746284607.031991094] [intermittent_talker]: Publishing: "Hello World: 13"
[INFO] [1746284607.531893032] [intermittent_talker]: Publishing: "Hello World: 14"
[INFO] [1746284608.031936173] [intermittent_talker]: Publishing: "Hello World: 15"
[INFO] [1746284608.531926245] [intermittent_talker]: Publishing: "Hello World: 16"
[INFO] [1746284609.031952104] [intermittent_talker]: Publishing: "Hello World: 17"
[INFO] [1746284609.531907729] [intermittent_talker]: Publishing: "Hello World: 18"
[INFO] [1746284610.031896174] [intermittent_talker]: Publishing: "Hello World: 19"
[INFO] [1746284610.531915395] [intermittent_talker]: Publishing: "Hello World: 20"
[INFO] [1746284611.032094980] [intermittent_talker]: Publishing: "Hello World: 21"
[INFO] [1746284611.531943200] [intermittent_talker]: Publishing: "Hello World: 22"
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors custom_executor
[INFO] [1746284622.340934455] [talker]: Publishing: "Hello World: 0"
[INFO] [1746284622.341753738] [listener]: I heard: "Hello World: 0"
[INFO] [1746284623.332281365] [talker]: Publishing: "Hello World: 1"
[INFO] [1746284623.333128730] [listener]: I heard: "Hello World: 1"
[INFO] [1746284624.332314051] [talker]: Publishing: "Hello World: 2"
[INFO] [1746284624.333083624] [listener]: I heard: "Hello World: 2"
[INFO] [1746284625.332371385] [talker]: Publishing: "Hello World: 3"
[INFO] [1746284625.333183938] [listener]: I heard: "Hello World: 3"
[INFO] [1746284626.332408452] [talker]: Publishing: "Hello World: 4"
[INFO] [1746284626.333192034] [listener]: I heard: "Hello World: 4"
[INFO] [1746284627.332296607] [talker]: Publishing: "Hello World: 5"
[INFO] [1746284627.333107608] [listener]: I heard: "Hello World: 5"
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors listener
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# ros2 run examples_rclpy_executors listener &
[1] 367290
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclpy_executors talker
[INFO] [1746284650.102183366] [talker]: Publishing: "Hello World: 0"
[INFO] [1746284650.110701972] [listener]: I heard: "Hello World: 0"
[INFO] [1746284651.094222758] [talker]: Publishing: "Hello World: 1"
[INFO] [1746284651.094998773] [listener]: I heard: "Hello World: 1"
[INFO] [1746284652.094060638] [talker]: Publishing: "Hello World: 2"
[INFO] [1746284652.094669842] [listener]: I heard: "Hello World: 2"
^C[ros2run]: Received signal:  Interrupt
root@cluttered:~# uname -m
x86_64
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_cyclonedds_cpp
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde