ros2/kilted_tutorial_party #189: Executables in `examples_rclpy_executors`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors callback_group
       ```
  
    
  </details>
- [ ] **Check `composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors composed
       ```
  
    
  </details>
- [ ] **Check `custom_callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_callback_group
       ```
  
    
  </details>
- [ ] **Check `custom_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_executor
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors listener
       ```
  
    
  </details>
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Noel215, 2025-05-03:
All are working fine.

---
Set up:

```
noel@laptop:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
noel@laptop:~$ dpkg --print-architecture
amd64
noel@laptop:~$ printenv AMENT_PREFIX_PATH 
/opt/ros/kilted
noel@laptop:~$ printenv RMW_IMPLEMENTATION
noel@laptop:~$ 
```
---

**Check `callback_group`**
```
noel@laptop:~$ ros2 run examples_rclpy_executors callback_group
[INFO] [1746311645.505788003] [double_talker]: Publishing: "Hello World: 0"
[INFO] [1746311645.507767909] [listener]: I heard: "Hello World: 0"
[INFO] [1746311645.963560833] [double_talker]: Publishing: "Hello World: 1"
[INFO] [1746311645.970683173] [double_talker]: Publishing: "Hello World: 2"
[INFO] [1746311645.972191233] [listener]: I heard: "Hello World: 1"
[INFO] [1746311645.973270851] [listener]: I heard: "Hello World: 2"
[INFO] [1746311646.459870116] [double_talker]: Publishing: "Hello World: 3"
[INFO] [1746311646.461080119] [listener]: I heard: "Hello World: 3"
[INFO] [1746311646.960616414] [double_talker]: Publishing: "Hello World: 4"
[INFO] [1746311646.964178108] [double_talker]: Publishing: "Hello World: 5"
[INFO] [1746311646.965062527] [listener]: I heard: "Hello World: 4"
[INFO] [1746311646.970414790] [listener]: I heard: "Hello World: 5"
[INFO] [1746311647.462817281] [double_talker]: Publishing: "Hello World: 6"
[INFO] [1746311647.470129157] [listener]: I heard: "Hello World: 6"
[...]
```
---
**Check `composed`**
```
noel@laptop:~$ ros2 run examples_rclpy_executors composed
[INFO] [1746311737.931759831] [talker]: Publishing: "Hello World: 0"
[INFO] [1746311737.932405627] [listener]: I heard: "Hello World: 0"
[INFO] [1746311738.921014009] [talker]: Publishing: "Hello World: 1"
[INFO] [1746311738.921550383] [listener]: I heard: "Hello World: 1"
[INFO] [1746311739.922873124] [talker]: Publishing: "Hello World: 2"
[INFO] [1746311739.925295381] [listener]: I heard: "Hello World: 2"
[INFO] [1746311740.922843239] [talker]: Publishing: "Hello World: 3"
[INFO] [1746311740.925260646] [listener]: I heard: "Hello World: 3"
[INFO] [1746311741.922865774] [talker]: Publishing: "Hello World: 4"
[INFO] [1746311741.923743789] [listener]: I heard: "Hello World: 4"
[...]
```
---
**Check `custom_callback_group`**
```
noel@laptop:~$ ros2 run examples_rclpy_executors custom_callback_group
[INFO] [1746311800.933984436] [intermittent_talker]: Publishing: "Hello World: 0"
[INFO] [1746311801.018224656] [intermittent_talker]: Publishing: "Hello World: 1"
[INFO] [1746311801.118149130] [intermittent_talker]: Publishing: "Hello World: 2"
[INFO] [1746311801.218018758] [intermittent_talker]: Publishing: "Hello World: 3"
[INFO] [1746311801.318471861] [intermittent_talker]: Publishing: "Hello World: 4"
[INFO] [1746311801.418137219] [intermittent_talker]: Publishing: "Hello World: 5"
[INFO] [1746311801.518125374] [intermittent_talker]: Publishing: "Hello World: 6"
[INFO] [1746311801.618123153] [intermittent_talker]: Publishing: "Hello World: 7"
[INFO] [1746311801.718106511] [intermittent_talker]: Publishing: "Hello World: 8"
[INFO] [1746311801.818472529] [intermittent_talker]: Publishing: "Hello World: 9"
[... 30 lines cut ...]
```
---
**Check `listener`**
```
noel@laptop:~$ ros2 run examples_rclpy_executors listener
[INFO] [1746312064.496993126] [listener]: I heard: "Hello World: 0"
[INFO] [1746312065.478718404] [listener]: I heard: "Hello World: 1"
[INFO] [1746312066.478229324] [listener]: I heard: "Hello World: 2"
[INFO] [1746312067.478066123] [listener]: I heard: "Hello World: 3"
[INFO] [1746312068.475470226] [listener]: I heard: "Hello World: 4"
[INFO] [1746312069.477845306] [listener]: I heard: "Hello World: 5"
[...]
```
---
**Check `talker`**
```
noel@laptop:~$ ros2 run examples_rclpy_executors talker
[INFO] [1746312064.486942464] [talker]: Publishing: "Hello World: 0"
[INFO] [1746312065.475735206] [talker]: Publishing: "Hello World: 1"
[INFO] [1746312066.475656548] [talker]: Publishing: "Hello World: 2"
[INFO] [1746312067.475625559] [talker]: Publishing: "Hello World: 3"
[INFO] [1746312068.474487958] [talker]: Publishing: "Hello World: 4"
[INFO] [1746312069.475633305] [talker]: Publishing: "Hello World: 5"
[...]
```

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde