ros2/kilted_tutorial_party #190: Executables in `examples_rclpy_executors`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors callback_group
       ```
  
    
  </details>
- [ ] **Check `composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors composed
       ```
  
    
  </details>
- [ ] **Check `custom_callback_group`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_callback_group
       ```
  
    
  </details>
- [ ] **Check `custom_executor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors custom_executor
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors listener
       ```
  
    
  </details>
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_executors talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

Check callback_group:
[image]

Check composed:
[image]

Check custom_callback_group:
[image]

Check custom_executor:
[image]

Check talker and listener (I killed the listener before the talker, hence why it publishes more messages that are not seen on the listener terminal):
[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde