ros2/kilted_tutorial_party #191: Executables in `examples_rclpy_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_cancel
       ```
  
    
  </details>
- [ ] **Check `client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_not_composable
       ```
  
    
  </details>
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed all checks for Issue #191 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run examples_rclpy_minimal_action_client client_cancel`. Node started successfully, printed "Waiting for action server...", and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_client client_not_composable`. Node started successfully, printed "Waiting for action server...", and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_client client`. Node started successfully, printed "Waiting for action server...", and ran without errors.

Additionally, verified that `examples_rclpy_minimal_action_client client` interacts correctly (sends goal, gets feedback, receives result) when `examples_rclpy_minimal_action_server server` is running in a separate terminal.

All executables started and functioned as expected.

closed 2025-05-05 by ahcorde