ros2/kilted_tutorial_party #192: Executables in `examples_rclpy_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `client_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_cancel
       ```
  
    
  </details>
- [ ] **Check `client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_not_composable
       ```
  
    
  </details>
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-04:
I am interested.

Tenveis, 2025-05-04:
- [X] Check `client_cancel`
- [X] Check `client_not_composable`
- [X] Check `client`

Following is the output when I followed the instructions.
[image]

I ran the respective server node to get the output.

- [X] Check `client_cancel`
[image]

- [X] Check `client_not_composable`
[image]

- [X] Check `client`
[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde