ros2/kilted_tutorial_party #193: Executables in `examples_rclpy_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_cancel
       ```
  
    
  </details>
- [ ] **Check `client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_not_composable
       ```
  
    
  </details>
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine.**

Check `client_cancel`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_client client_cancel
[INFO] [1746227868.153414400] [minimal_action_client]: Waiting for action server...
[INFO] [1746227868.155500000] [minimal_action_client]: Sending goal request...
[INFO] [1746227868.220597900] [minimal_action_client]: Goal accepted :)
[INFO] [1746227868.225386700] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1746227869.227327700] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227870.226498000] [minimal_action_client]: Canceling goal
[INFO] [1746227870.232912600] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227870.235409600] [minimal_action_client]: Goal successfully canceled
```

Check `client_not_composable`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_client client_not_composable
[INFO] [1746227902.804588100] [minimal_action_client]: Waiting for action server...
[INFO] [1746227902.807243100] [minimal_action_client]: Sending goal request...
[INFO] [1746227902.868947200] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1746227902.870509900] [minimal_action_client]: Goal accepted :)
[INFO] [1746227903.821012200] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227904.825709800] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227905.829509300] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227906.832353600] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227907.835068500] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227908.837618100] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227909.841442500] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227910.845550800] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227911.850305400] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Check `client`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_client client
[INFO] [1746227935.113087100] [minimal_action_client]: Waiting for action server...
[INFO] [1746227935.115636500] [minimal_action_client]: Sending goal request...
[INFO] [1746227935.198145800] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1746227935.201153200] [minimal_action_client]: Goal accepted :)
[INFO] [1746227936.128022600] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227937.130498300] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227938.133625900] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227939.139830600] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227940.142309000] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227941.145760400] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227942.151065800] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227943.153161100] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227944.158600600] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde