ros2/kilted_tutorial_party #194: Executables in `examples_rclpy_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `client_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_cancel
       ```
  
    
  </details>
- [ ] **Check `client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client_not_composable
       ```
  
    
  </details>
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_client client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Noel215, 2025-05-03:
Hi, could you assign me this one?

Noel215, 2025-05-03:
All of them LGTM.

---
**Check `client_cancel`:**

Server:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_server server
[INFO] [1746310644.032713287] [minimal_action_server]: Received goal request
[INFO] [1746310644.033542235] [minimal_action_server]: Executing goal...
[INFO] [1746310644.033885778] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746310645.035948206] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746310646.036216740] [minimal_action_server]: Received cancel request
[INFO] [1746310646.037288737] [minimal_action_server]: Goal canceled
```
Client:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_client client_cancel
[INFO] [1746310762.179143617] [minimal_action_client]: Waiting for action server...
[INFO] [1746310762.681497660] [minimal_action_client]: Sending goal request...
[INFO] [1746310762.699765188] [minimal_action_client]: Goal accepted :)
[INFO] [1746310762.700531450] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1746310763.702104522] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1746310764.702168073] [minimal_action_client]: Canceling goal
[INFO] [1746310764.705798647] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746310764.706852224] [minimal_action_client]: Goal successfully canceled
```
---
 **Check `client_not_composable`**

Server:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_server server
[INFO] [1746310775.807959030] [minimal_action_server]: Received goal request
[INFO] [1746310775.808904012] [minimal_action_server]: Executing goal...
[INFO] [1746310775.809854731] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746310776.811924425] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746310777.814356909] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746310778.816774013] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746310779.819134520] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746310780.821539763] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746310781.824004400] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746310782.826331816] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746310783.828681777] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746310784.832209935] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Client:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_client client_not_composable
[INFO] [1746310775.538522560] [minimal_action_client]: Waiting for action server...
[INFO] [1746310775.790701904] [minimal_action_client]: Sending goal request...
[INFO] [1746310775.808909678] [minimal_action_client]: Goal accepted :)
[INFO] [1746310775.810338881] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1746310776.814397797] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1746310777.816905737] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746310778.819229454] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746310779.821597047] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746310780.824038957] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746310781.826497809] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746310782.828863239] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746310783.831199736] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746310784.836486785] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```
---
**Check `client`**

Server:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_server server
[... 6 lines cut ...]
[INFO] [1746310802.008243216] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746310803.008868726] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746310804.010592490] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746310805.011373884] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746310806.013513283] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746310807.014577103] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Client:
```
noel@laptop:~$ ros2 run examples_rclpy_minimal_action_client client
[INFO] [1746310797.741464638] [minimal_action_client]: Waiting for action server...
[INFO] [1746310797.992095869] [minimal_action_client]: Sending goal request...
[INFO] [1746310798.004386535] [minimal_action_client]: Goal accepted :)
[... cut ...]
[INFO] [1746310803.009631646] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746310804.012482263] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746310805.012121145] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746310806.016081451] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746310807.015527272] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Noel215, 2025-05-03:
Set up:
```
noel@laptop:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
noel@laptop:~$ dpkg --print-architecture
amd64
noel@laptop:~$ printenv AMENT_PREFIX_PATH 
/opt/ros/kilted
noel@laptop:~$ printenv RMW_IMPLEMENTATION
noel@laptop:~$ 
```

closed 2025-05-09 by mjcarroll