ros2/kilted_tutorial_party #196: Executables in `examples_rclpy_minimal_action_server`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `server_defer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_defer
       ```
  
    
  </details>
- [ ] **Check `server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_not_composable
       ```
  
    
  </details>
- [ ] **Check `server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server
       ```
  
    
  </details>
- [ ] **Check `server_queue_goals`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_queue_goals
       ```
  
    
  </details>
- [ ] **Check `server_single_goal`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_single_goal
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed all checks for Issue #196 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run examples_rclpy_minimal_action_server server_defer`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_server server_not_composable`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_server server`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_server server_queue_goals`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_action_server server_single_goal`. Node started successfully and ran without errors.

Additionally, verified that the basic `server` executable interacts correctly with `examples_rclpy_minimal_action_client client`, receiving the goal, sending feedback, and returning the result as expected.

(Noted a minor `[ERROR] [rcl]: Failed to remove timer jump callback` message appears only upon stopping nodes with Ctrl+C, but runtime functionality was correct).

All action server examples started and functioned as expected during runtime.

closed 2025-05-09 by mjcarroll