ros2/kilted_tutorial_party #197: Executables in `examples_rclpy_minimal_action_server`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `server_defer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_defer
       ```
  
    
  </details>
- [ ] **Check `server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_not_composable
       ```
  
    
  </details>
- [ ] **Check `server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server
       ```
  
    
  </details>
- [ ] **Check `server_queue_goals`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_queue_goals
       ```
  
    
  </details>
- [ ] **Check `server_single_goal`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_single_goal
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-04:
I am interested.

Tenveis, 2025-05-04:
I ran the respective `client` node to get output.

- [X] Check `server_defer`
[image]

- [X] Check `server_not_composable`
[image]

- [X] Check `server`
[image]

- [X] Check `server_queue_goals`
[image]

- [X] Check `server_single_goal
[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde