ros2/kilted_tutorial_party #198: Executables in `examples_rclpy_minimal_action_server`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `server_defer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_defer
       ```
  
    
  </details>
- [ ] **Check `server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_not_composable
       ```
  
    
  </details>
- [ ] **Check `server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server
       ```
  
    
  </details>
- [ ] **Check `server_queue_goals`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_queue_goals
       ```
  
    
  </details>
- [ ] **Check `server_single_goal`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_single_goal
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine.**

Check `server_defer`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_server server_defer
[INFO] [1746227585.254805600] [minimal_action_server]: Received goal request
[INFO] [1746227585.259022600] [minimal_action_server]: Deferring execution...
[INFO] [1746227588.269351100] [minimal_action_server]: Executing goal...
[INFO] [1746227588.272144600] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746227589.274349200] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227590.278682500] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227591.283279000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227592.286812000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227593.291829500] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227594.294151100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227595.297569000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227596.301281000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227597.305038600] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Check `server_not_composable`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_server server_not_composable
[INFO] [1746227632.015325900] [minimal_action_server]: Executing goal...
[INFO] [1746227632.019278000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746227633.022113000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227634.026898400] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227635.030047000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227636.034255900] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227637.039707000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227638.043386700] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227639.047211900] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227640.050408700] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227641.054981700] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Check `server`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_server server
[INFO] [1746227674.240033000] [minimal_action_server]: Received goal request
[INFO] [1746227674.244020100] [minimal_action_server]: Executing goal...
[INFO] [1746227674.244911300] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746227675.247239100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227676.251280600] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227677.253464200] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227678.258203100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227679.261104600] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227680.264117700] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227681.268496200] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227682.272921100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227683.275052000] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

Check `server_queue_goals`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_action_server server_queue_goals
[INFO] [1746227716.474000800] [minimal_action_server]: Received goal request
[INFO] [1746227716.479714400] [minimal_action_server]: Executing goal...
[INFO] [1746227716.482765700] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1746227717.484849000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1746227718.487990800] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1746227719.493687300] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1746227720.497518000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1746227721.502252000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227722.506076100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227723.510221800] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227724.512589000] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[... cut ...]
[INFO] [1746227759.895370300] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1746227760.899188400] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1746227761.901923100] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1746227762.905511200] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1746227763.910246200] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde