ros2/kilted_tutorial_party #2: Adding A Frame Cpp

tfoote, 2025-05-01:
Check the documentation for the 'Adding A Frame Cpp' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Adding A Frame Cpp page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Adding-A-Frame-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-03:
Hi I will try this

Aniketkumarroy, 2025-05-03:
ran the test and here the results
### **Adding a fixed frame**

[image]
### **following the fixed frame**

[image]
### **following a dynamic frame**

[image]

- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde