ros2/kilted_tutorial_party #20: Check Lifecyle node py

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Lifecycle Py](https://github.com/ros2/demos/tree/kilted/lifecycle_py)

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run lifecycle_py lifecycle_talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle nodes
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 1
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 3
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1680583079.479073069] [lc_talker]: on_configure() is called.
       [INFO] [1680583080.439717509] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
       ```
    1.
       ```bash
       # stdout in terminal 2
       /lc_talker
       ```
    1.
       ```bash
       # stdout in terminal 2
       Transitioning successful
       ```
    1.
       ```bash
       # stdout in terminal 2
       [INFO] [1680583403.113550155] [lc_talker]: on_activate() is called.
       [INFO] [1680583403.439875193] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #323]
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-06:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / Fast DDS.

ShahazadAbdulla, 2025-05-06:
Successfully completed checks for Issue #20 on Windows 11 / amd64 / binary (via Pixi) / Fast DDS.

- Ran `ros2 run lifecycle_py lifecycle_talker` in T1.
- Verified node discovery using `ros2 node list` and lifecycle service discovery using `ros2 service list | findstr lc_talker` in T2.
- Successfully transitioned the node using `ros2 lifecycle set /lc_talker configure` in T2. Observed correct `on_configure()` log and "publisher is inactive" messages in T1.
- Successfully transitioned the node using `ros2 lifecycle set /lc_talker activate` in T2. Observed correct `on_activate()` log and subsequent "Publishing: [Lifecycle HelloWorld #X]" messages in T1.

The Python lifecycle demo node and `ros2 lifecycle` CLI tool worked as expected.

ShahazadAbdulla, 2025-05-06:
Note: On Check listener and talker ii, the command should be '''ros2 node list''' not 'ros2 lifecycle node' right? or is it right?

Yadunund, 2025-05-09:
Thanks for testing! 

> Note: On Check listener and talker ii, the command should be '''ros2 node list''' not 'ros2 lifecycle node' right? or is the command 'ros2 lifecycle node' correct?

`ros2 lifecycle nodes` is the correctly cli command to only list nodes that are LifecyceNodes. `ros2 node list` will list all nodes including LifecycleNodes.

closed 2025-05-09 by Yadunund

ShahazadAbdulla, 2025-05-10:
Thanks for the clarification @Yadunund 

Just to confirm, I re-ran the `lifecycle_py lifecycle_talker` and then used `ros2 lifecycle nodes` in another terminal. It correctly listed `/lc_talker` as expected. So that discovery mechanism is also working well on Windows 11 / binary (Pixi) / Fast DDS.