ros2/kilted_tutorial_party #201: Executables in `examples_rclpy_minimal_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client
       ```
  
    
  </details>
- [ ] **Check `client_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async
       ```
  
    
  </details>
- [ ] **Check `client_async_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_member_function
       ```
  
    
  </details>
- [ ] **Check `client_async_callback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_callback
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed all checks for Issue #201 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

- Ran `ros2 run examples_rclpy_minimal_client client`. Node started successfully.
- Ran `ros2 run examples_rclpy_minimal_client client_async`. Node started successfully.
- Ran `ros2 run examples_rclpy_minimal_client client_async_member_function`. Node started successfully.
- Ran `ros2 run examples_rclpy_minimal_client client_async_callback`. Node started successfully.

Additionally, verified that all four client executables interact correctly with the `demo_nodes_py add_two_ints_server` running in a separate terminal. Each client successfully called the service and printed the correct result (e.g., "Result of add_two_ints: for 41 + 1 = 42").

All Python service client examples ran as expected.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde