ros2/kilted_tutorial_party #203: Executables in `examples_rclpy_minimal_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client
       ```
  
    
  </details>
- [ ] **Check `client_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async
       ```
  
    
  </details>
- [ ] **Check `client_async_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_member_function
       ```
  
    
  </details>
- [ ] **Check `client_async_callback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_callback
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine.**

Check `client`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_client client
[INFO] [1746227246.935658200] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
```

Check `client_async`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_client client_async
[INFO] [1746227325.559749800] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
```

Check `client_async_member_function`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_client client_async_member_function
[INFO] [1746227367.444437400] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
```

Check `client_async_callback`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_client client_async_callback
[INFO] [1746227398.633373700] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde