ros2/kilted_tutorial_party #206: Executables in `examples_rclpy_minimal_publisher`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `publisher_local_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_publisher publisher_local_function
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_old_school`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_publisher publisher_old_school
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-02:
I'll test this on Windows 11 / amd64 / binary (Pixi) / CycloneDDS.

ShahazadAbdulla, 2025-05-02:
Successfully completed all checks for Issue #206 on Windows 11 / amd64 / binary (Pixi) / CycloneDDS.

This test verifies that different basic implementations of rclpy publishers run correctly.

- `ros2 run examples_rclpy_minimal_publisher publisher_local_function`: Ran successfully and published messages.
- `ros2 run examples_rclpy_minimal_publisher publisher_member_function`: Ran successfully and published messages.
- `ros2 run examples_rclpy_minimal_publisher publisher_old_school`: Ran successfully and published messages.

All executables started publishing "Hello World: X" messages as expected.

Example verification output:

(pixi_ros2_rolling) C:\pixi_ws>ros2 run examples_rclpy_minimal_publisher publisher_local_function
[INFO] [1746216115.491850700] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1746216115.922213900] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1746216116.423523300] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1746216116.923521600] [minimal_publisher]: Publishing: "Hello World: 3"
[INFO] [1746216117.423254500] [minimal_publisher]: Publishing: "Hello World: 4"
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>ros2 run examples_rclpy_minimal_publisher publisher_member_function
[INFO] [1746216123.669837300] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1746216124.098888500] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1746216124.599249400] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1746216125.098860600] [minimal_publisher]: Publishing: "Hello World: 3"
[INFO] [1746216125.603715100] [minimal_publisher]: Publishing: "Hello World: 4"
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>ros2 run examples_rclpy_minimal_publisher publisher_old_school
[INFO] [1746216136.265112900] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1746216136.767554300] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1746216137.269922500] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1746216137.771881600] [minimal_publisher]: Publishing: "Hello World: 3"
[ros2run]: Received signal:  Interrupt

Everything worked as expected.

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde