ros2/kilted_tutorial_party #211: Executables in `examples_rclpy_minimal_service`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `service`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_service service
       ```
  
    
  </details>
- [ ] **Check `service_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_service service_member_function
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed all checks for Issue #211 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

- Ran `ros2 run examples_rclpy_minimal_service service`. Node started successfully, printed "Ready...", and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_service service_member_function`. Node started successfully, printed "Ready...", and ran without errors.

Additionally, verified that `service_member_function` interacts correctly with `examples_rclpy_minimal_client client_async`. The server received the request (logged "Incoming request a: 41 b: 1") and the client received the correct result.

Both Python service server examples ran as expected.

ahcorde, 2025-05-05:
Thank you for testing