ros2/kilted_tutorial_party #213: Executables in `examples_rclpy_minimal_service`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `service`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_service service
       ```
  
    
  </details>
- [ ] **Check `service_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_service service_member_function
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine.**

Check `service`

Output
```
C:\pixi_ws>ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{'a':2, 'b':2}"
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=2, b=2)

response:
example_interfaces.srv.AddTwoInts_Response(sum=4)
```

Check `service_member_function`

Output
```
C:\pixi_ws>ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{'a':2, 'b':4}"
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=2, b=4)

response:
example_interfaces.srv.AddTwoInts_Response(sum=6)
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde