ros2/kilted_tutorial_party #216: Executables in `examples_rclpy_minimal_subscriber`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_old_school`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed all checks for Issue #216 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

- Ran `ros2 run examples_rclpy_minimal_subscriber subscriber_lambda`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_subscriber subscriber_member_function`. Node started successfully and ran without errors.
- Ran `ros2 run examples_rclpy_minimal_subscriber subscriber_old_school`. Node started successfully and ran without errors.

Additionally, verified that the subscribers interact correctly (receive "I heard..." messages) when the publisher `examples_rclcpp_minimal_publisher publisher_member_function` is running in a separate terminal. 
But when trying to run the  corresponding Python publisher `examples_rclpy_minimal_publisher publisher_member_function` is `No executable found`

All Python subscriber examples started and functioned as expected.

ahcorde, 2025-05-05:
Thank you for testing