ros2/kilted_tutorial_party #218: Executables in `examples_rclpy_minimal_subscriber`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_old_school`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine**

I sent Hi msgs using `ros2 topic pub`

Check `subscriber_lambda`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
[INFO] [1746226073.047138700] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226074.025185400] [minimal_subscriber]: I heard: "Hi"
```

check `subscriber_member_function`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
[INFO] [1746226196.115303200] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226197.081353600] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226198.080533800] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226199.080951300] [minimal_subscriber]: I heard: "Hi"
```

check `subscriber_old_school`

```
C:\pixi_ws>ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
[INFO] [1746226230.133969800] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226231.093759900] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226232.093011700] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226233.093296800] [minimal_subscriber]: I heard: "Hi"
[INFO] [1746226234.094248300] [minimal_subscriber]: I heard: "Hi"
```

ahcorde, 2025-05-05:
Thank yo for testing

closed 2025-05-05 by ahcorde