ros2/kilted_tutorial_party #219: Executables in `examples_rclpy_minimal_subscriber`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_old_school`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-02:
Setup:
`legends@4legends-ASUS:~/rviz2_ws/src/my-pkg$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted
`
`legends@4legends-ASUS:~/rviz2_ws/src/my-pkg$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
`

- Check subscriber_lambda
`legends@4legends-ASUS:~$ ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
[INFO] [1746173932.368755181] [minimal_subscriber]: I heard: "Hello, world! 0"
[INFO] [1746173932.859566179] [minimal_subscriber]: I heard: "Hello, world! 1"
[INFO] [1746173933.359625402] [minimal_subscriber]: I heard: "Hello, world! 2"
[INFO] [1746173933.859564804] [minimal_subscriber]: I heard: "Hello, world! 3"
[INFO] [1746173934.359837054] [minimal_subscriber]: I heard: "Hello, world! 4"
[INFO] [1746173934.859829515] [minimal_subscriber]: I heard: "Hello, world! 5"
[INFO] [1746173935.359698024] [minimal_subscriber]: I heard: "Hello, world! 6"
[INFO] [1746173935.859607724] [minimal_subscriber]: I heard: "Hello, world! 7"
^C[ros2run]: Received signal:  Interrupt
`
- Check subscriber_member_function
`legends@4legends-ASUS:~$ ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
[INFO] [1746174062.423553264] [minimal_subscriber]: I heard: "Hello, world! 6"
[INFO] [1746174062.913828450] [minimal_subscriber]: I heard: "Hello, world! 7"
[INFO] [1746174063.413512325] [minimal_subscriber]: I heard: "Hello, world! 8"
[INFO] [1746174063.913676302] [minimal_subscriber]: I heard: "Hello, world! 9"
[INFO] [1746174064.413665651] [minimal_subscriber]: I heard: "Hello, world! 10"
[INFO] [1746174064.913730131] [minimal_subscriber]: I heard: "Hello, world! 11"
[INFO] [1746174065.413113240] [minimal_subscriber]: I heard: "Hello, world! 12"
[INFO] [1746174065.913650054] [minimal_subscriber]: I heard: "Hello, world! 13"
[INFO] [1746174066.413514642] [minimal_subscriber]: I heard: "Hello, world! 14"
^C[ros2run]: Received signal:  Interrupt
`
- Check subscriber_old_school
`legends@4legends-ASUS:~$ ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
[INFO] [1746174100.924042477] [minimal_subscriber]: I heard: "Hello, world! 83"
[INFO] [1746174101.413423343] [minimal_subscriber]: I heard: "Hello, world! 84"
[INFO] [1746174101.913416806] [minimal_subscriber]: I heard: "Hello, world! 85"
[INFO] [1746174102.413169256] [minimal_subscriber]: I heard: "Hello, world! 86"
[INFO] [1746174102.913308760] [minimal_subscriber]: I heard: "Hello, world! 87"
[INFO] [1746174103.413597652] [minimal_subscriber]: I heard: "Hello, world! 88"
[INFO] [1746174103.914105167] [minimal_subscriber]: I heard: "Hello, world! 89"
[INFO] [1746174104.413108838] [minimal_subscriber]: I heard: "Hello, world! 90"
[INFO] [1746174104.913742245] [minimal_subscriber]: I heard: "Hello, world! 91"
[INFO] [1746174105.413149361] [minimal_subscriber]: I heard: "Hello, world! 92"
[INFO] [1746174105.913354401] [minimal_subscriber]: I heard: "Hello, world! 93"
[INFO] [1746174106.413469536] [minimal_subscriber]: I heard: "Hello, world! 94"
^C[ros2run]: Received signal:  Interrupt
`

ahcorde, 2025-05-02:
Thank you for testing!

closed 2025-05-02 by ahcorde