ros2/kilted_tutorial_party #221: Executables in `lifecycle`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifecycle_demo_launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch lifecycle lifecycle_demo_launch.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed the check for Issue #221 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

- Ran `ros2 launch lifecycle lifecycle_demo_launch.py`.
- The launch file started successfully.
- Log output showed the `lc_talker`, `lc_listener`, and `lc_client` nodes starting and interacting.
- Observed messages indicating successful state transitions (activating, publishing, deactivating, inactive) triggered by the lifecycle manager/client.
- The talker correctly published messages when active and stopped when inactive.

The lifecycle demo launch file executed as expected.
(Noted a common warning on shutdown: `LifecycleNode is not shut down... Node still in state(2)...` after Ctrl+C).

closed 2025-05-05 by ahcorde