ros2/kilted_tutorial_party #223: Executables in `lifecycle`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifecycle_demo_launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch lifecycle lifecycle_demo_launch.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
The executables are working fine.

```
C:\pixi_ws>ros2 launch lifecycle lifecycle_demo_launch.py
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-15-29-03-346597-DESKTOP-8BMD1U3-5944' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-15-29-03-346597-DESKTOP-8BMD1U3-5944
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_talker.EXE-1]: process started with pid [5664]
[INFO] [lifecycle_listener.EXE-2]: process started with pid [8456]
[INFO] [lifecycle_service_client.EXE-3]: process started with pid [1964]
[lifecycle_talker.EXE-1] [INFO] [1746224943.947684000] [lc_talker]: on_configure() is called.
[lifecycle_listener.EXE-2] [INFO] [1746224943.948419200] [lc_listener]: notify callback: Transition from state configuring to inactive
[lifecycle_service_client.EXE-3] [INFO] [1746224943.948571100] [lc_client]: Transition 1 successfully triggered.
[lifecycle_service_client.EXE-3] [INFO] [1746224943.949009200] [lc_client]: Node lc_talker has current state inactive.
[lifecycle_talker.EXE-1] [INFO] [1746224944.956190700] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [WARN] [1746224944.956242300] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[lifecycle_talker.EXE-1] [INFO] [1746224945.956128800] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224946.955581300] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224947.971820400] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224948.956587500] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224949.956896600] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224950.956718000] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224951.956104500] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224952.956112800] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker.EXE-1] [INFO] [1746224953.784913600] [lc_talker]: on_activate() is called.
[lifecycle_listener.EXE-2] [INFO] [1746224953.784997200] [lc_listener]: notify callback: Transition from state inactive to activating
[lifecycle_listener.EXE-2] [INFO] [1746224955.785549200] [lc_listener]: notify callback: Transition from state activating to active
[lifecycle_service_client.EXE-3] [INFO] [1746224955.785621500] [lc_client]: Transition 3 successfully triggered.
[lifecycle_talker.EXE-1] [INFO] [1746224955.785680100] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #10]
[lifecycle_listener.EXE-2] [INFO] [1746224955.785913200] [lc_listener]: data_callback: Lifecycle HelloWorld #10
[lifecycle_service_client.EXE-3] [INFO] [1746224955.785995400] [lc_client]: Node lc_talker has current state active.
[lifecycle_talker.EXE-1] [INFO] [1746224955.956461500] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #11]
[lifecycle_listener.EXE-2] [INFO] [1746224955.956689000] [lc_listener]: data_callback: Lifecycle HelloWorld #11
[lifecycle_talker.EXE-1] [INFO] [1746224956.956582600] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #12]
[lifecycle_listener.EXE-2] [INFO] [1746224956.956886100] [lc_listener]: data_callback: Lifecycle HelloWorld #12
[lifecycle_talker.EXE-1] [INFO] [1746224957.956354600] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #13]
[lifecycle_listener.EXE-2] [INFO] [1746224957.956650300] [lc_listener]: data_callback: Lifecycle HelloWorld #13
[lifecycle_talker.EXE-1] [INFO] [1746224958.956161400] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #14]
[lifecycle_listener.EXE-2] [INFO] [1746224958.956481200] [lc_listener]: data_callback: Lifecycle HelloWorld #14
[lifecycle_talker.EXE-1] [INFO] [1746224959.957537400] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #15]
[lifecycle_listener.EXE-2] [INFO] [1746224959.958061800] [lc_listener]: data_callback: Lifecycle HelloWorld #15
[lifecycle_talker.EXE-1] [INFO] [1746224960.956894900] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #16]
[lifecycle_listener.EXE-2] [INFO] [1746224960.957137600] [lc_listener]: data_callback: Lifecycle HelloWorld #16
[lifecycle_talker.EXE-1] [INFO] [1746224961.957169200] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #17]
[lifecycle_listener.EXE-2] [INFO] [1746224961.957386500] [lc_listener]: data_callback: Lifecycle HelloWorld #17
[lifecycle_talker.EXE-1] [INFO] [1746224962.956530200] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #18]
[... cut ...]
[WARNING] [lifecycle_talker.EXE-1]: 'SIGINT' sent to process[lifecycle_talker.EXE-1] not supported on Windows, escalating to 'SIGTERM'
[INFO] [lifecycle_talker.EXE-1]: sending signal 'SIGTERM' to process[lifecycle_talker.EXE-1]
[ERROR] [lifecycle_listener.EXE-2]: process has died [pid 8456, exit code 1, cmd 'C:\pixi_ws\ros2-windows\lib\lifecycle\lifecycle_listener.EXE --ros-args'].
[ERROR] [lifecycle_talker.EXE-1]: process has died [pid 5664, exit code 1, cmd 'C:\pixi_ws\ros2-windows\lib\lifecycle\lifecycle_talker.EXE --ros-args -r __node:=lc_talker -r __ns:=/'].
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde