ros2/kilted_tutorial_party #225: Executables in `lifecycle`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `lifecycle_demo_launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch lifecycle lifecycle_demo_launch.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

The transition seems to occur as expected:
[image]

Not sure if the output after the SIGINT is what is expected, but everything else seems fine:
[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde