ros2/kilted_tutorial_party #226: Executables in `pendulum_control`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `pendulum_demo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo
       ```
  
    
  </details>
- [ ] **Check `pendulum_teleop`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_teleop
       ```
  
    
  </details>
- [ ] **Check `pendulum_logger`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_logger
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-04:
- Setup
```
legends@4legends-ASUS:~$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

legends@4legends-ASUS:~$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

``` 

- [x] Check pendulum_demo
```
legends@4legends-ASUS:~$ ros2 run pendulum_control pendulum_demo
Couldn't set scheduling priority and policy: Operation not permitted
Caught exception: std::bad_alloc
Unlocking memory and continuing.
Couldn't lock all cached virtual memory.
Pagefaults from reading pages not yet mapped into RAM will be recorded.
Initial major pagefaults: 0
Initial minor pagefaults: 1910013
Pendulum set to: 1.570796
No results filename given, not writing results
rttest statistics:
  - Minor pagefaults: 0
  - Major pagefaults: 0
  Latency (time after deadline was missed):
    - Min: 53961 ns
    - Max: 243041 ns
    - Mean: 70730.870000 ns
    - Standard deviation: 14783.284981

PendulumMotor received 498 messages
PendulumController received 499 messages

``` 
- [x] Check pendulum_teleop

```
legends@4legends-ASUS:~$ ros2 run pendulum_control pendulum_teleop
Command argument not specified. Setting command to 90 degrees (PI/2 radians).
Teleop message published.
Teleop node exited.

``` 
- [x] Check pendulum_logger

```
Commanded motor angle: 1.570796
Actual motor angle: 1.520562
Current latency: 59776 ns
Mean latency: 70846.887712 ns
Min latency: 53961 ns
Max latency: 243041 ns
Minor pagefaults during execution: 0
Major pagefaults during execution: 0

Commanded motor angle: 1.570796
Actual motor angle: 1.470326
Current latency: 59244 ns
Mean latency: 70834.609524 ns
Min latency: 53961 ns
Max latency: 243041 ns
Minor pagefaults during execution: 0
Major pagefaults during execution: 0

Commanded motor angle: 1.570796
[... 456 lines cut ...]
Major pagefaults during execution: 0

Commanded motor angle: 1.570796
Actual motor angle: 1.470281
Current latency: 70244 ns
Mean latency: 70736.571715 ns
Min latency: 53961 ns
Max latency: 243041 ns
Minor pagefaults during execution: 0
Major pagefaults during execution: 0

Commanded motor angle: 1.570796
Actual motor angle: 1.520538
Current latency: 69397 ns
Mean latency: 70735.229459 ns
Min latency: 53961 ns
Max latency: 243041 ns
Minor pagefaults during execution: 0
Major pagefaults during execution: 0

^C[ros2run]: Received signal:  Interrupt
[INFO] [1746327356.273415872] [rclcpp]: signal_handler(signum=2)
[INFO] [1746327356.273567869] [rclcpp]: signal_handler(signum=2)

```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde