ros2/kilted_tutorial_party #228: Executables in `pendulum_control`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `pendulum_demo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo
       ```
  
    
  </details>
- [ ] **Check `pendulum_teleop`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_teleop
       ```
  
    
  </details>
- [ ] **Check `pendulum_logger`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_logger
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
### Checks

* [x]  **Check pendulum_demo**   
* [x]  **Check pendulum_teleop**   
* [x]  **Check pendulum_logger**   

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde