ros2/kilted_tutorial_party #229: Executables in `quality_of_service_demo_cpp`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed checks for Issue #299 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

Note: Running the executables without arguments only printed help messages. Testing was performed by adding example arguments (e.g., `lifespan 1000`, `liveliness 1000`, `deadline 600`).

- Ran `ros2 run quality_of_service_demo_cpp lifespan 1000`. Node started successfully and ran without errors, demonstrating lifespan behavior.
- Ran `ros2 run quality_of_service_demo_cpp liveliness 1000`. Node started successfully and ran without errors, demonstrating liveliness event callbacks.
- Ran `ros2 run quality_of_service_demo_cpp deadline 600`. Node started successfully and ran without errors, correctly demonstrating deadline missed events.

All C++ QoS demo executables appear functional when run with appropriate arguments. The issue description's "Checks" section might benefit from including example arguments.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde