ros2/kilted_tutorial_party #23: Check ROS 2 bag CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check record**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You should stop the talker once the recording is stopped.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
    1.
       Run for a few seconds and then hit Ctrl-C to stop the bag recording.
       ```bash
       # User input in terminal 2
       ros2 bag record /chatter -o my-bag
       ```
  
    **Expect**
    
    1.
       You should see a folder named my-bag in the current directory
    
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous check.
       
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info my-bag
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       Files:             my-bag_0.mcap
       Bag size:          3.0 KiB
       Storage id:        mcap
       Duration:          3.999s
       Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
       End:               Apr  2 2023 13:45:51.834 (1680414351.834)
       Messages:          5
       Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr
       ```
  </details>
- [ ] **Check play**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous two checks. Make sure that the talker is stopped.
       
    1.
       The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 bag play -l my-bag
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /chatter
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       data: 'Hello World: 77'
       ---
       data: 'Hello World: 78'
       ---
       data: 'Hello World: 75'
       ---
       data: 'Hello World: 76'
       ---
       data: 'Hello World: 77'
       ---
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-02:
I'll test this on Windows 11 / amd64 / binary (Pixi) / CycloneDDS.

ShahazadAbdulla, 2025-05-02:
Successfully completed all checks for Issue #23 on Windows 11 / amd64 / binary (Pixi) / CycloneDDS.

- **Check record:** Ran `ros2 run demo_nodes_cpp talker` in one terminal and `ros2 bag record /chatter -o my_bag` in another. Recording stopped successfully with Ctrl+C, and the `my_bag` folder was created.
- **Check info:** Ran `ros2 bag info my_bag`. Output showed the expected structure (file, size, duration, message count, topic info, etc.). 
- Example snippet:

(pixi_ros2_rolling) C:\pixi_ws>ros2 bag info my_bag

Files:             my_bag_0.mcap
Bag size:          6.7 KiB
Storage id:        mcap
ROS Distro:        kilted
Duration:          24.011992100s
Start:             May  3 2025 01:10:24.631953000 (1746214824.631953000)
End:               May  3 2025 01:10:48.643945100 (1746214848.643945100)
Messages:          25
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 25 | Serialization Format: cdr
Services:          0
Service information:
Actions:           0
Action information:

- **Check play:** Stopped the talker. Ran `ros2 bag play -l my_bag` in one terminal and `ros2 topic echo /chatter` in another. Messages from the bag file were successfully echoed in the second terminal, confirming playback.

Everything worked as expected.

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde