ros2/kilted_tutorial_party #231: Executables in `quality_of_service_demo_cpp`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine.**

Check `lifespan`

```
C:\pixi_ws>ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
[INFO] [1746225289.820151500] [talker]: Talker starting up
[INFO] [1746225289.824148100] [listener]: Listener starting up
[INFO] [1746225290.331024800] [talker]: Publishing: 'Talker says 0'
[INFO] [1746225290.830926500] [talker]: Publishing: 'Talker says 1'
[INFO] [1746225291.331446900] [talker]: Publishing: 'Talker says 2'
[INFO] [1746225291.831277300] [talker]: Publishing: 'Talker says 3'
[INFO] [1746225292.330601500] [talker]: Publishing: 'Talker says 4'
[INFO] [1746225292.831643600] [talker]: Publishing: 'Talker says 5'
[INFO] [1746225292.834871200] [listener]: Listener heard: [Talker says 4]
[INFO] [1746225292.835633000] [listener]: Listener heard: [Talker says 5]
[INFO] [1746225293.331325400] [talker]: Publishing: 'Talker says 6'
[INFO] [1746225293.333047400] [listener]: Listener heard: [Talker says 6]
[INFO] [1746225293.831454400] [talker]: Publishing: 'Talker says 7'
[INFO] [1746225293.833701200] [listener]: Listener heard: [Talker says 7]
[INFO] [1746225294.331432000] [talker]: Publishing: 'Talker says 8'
[INFO] [1746225294.333283300] [listener]: Listener heard: [Talker says 8]
[INFO] [1746225294.831473100] [talker]: Publishing: 'Talker says 9'
[INFO] [1746225294.833731200] [listener]: Listener heard: [Talker says 9]
```

Check `liveliness`

```
C:\pixi_ws>ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
[INFO] [1746225413.991519400] [talker]: Talker starting up
[INFO] [1746225413.995392400] [listener]: Listener starting up
[INFO] [1746225414.507846800] [talker]: Publishing: 'Talker says 0'
[INFO] [1746225414.508229800] [listener]: Listener heard: [Talker says 0]
[INFO] [1746225414.509924600] [listener]: Liveliness changed event:
[INFO] [1746225414.510331800] [listener]:   alive_count: 1
[INFO] [1746225414.510762800] [listener]:   not_alive_count: 0
[INFO] [1746225414.511055600] [listener]:   alive_count_change: 1
[INFO] [1746225414.511377900] [listener]:   not_alive_count_change: 0
[INFO] [1746225415.008172700] [talker]: Publishing: 'Talker says 1'
[INFO] [1746225415.009863100] [listener]: Listener heard: [Talker says 1]
[INFO] [1746225415.507751500] [talker]: Publishing: 'Talker says 2'
[INFO] [1746225415.510127000] [listener]: Listener heard: [Talker says 2]
[INFO] [1746225416.007783800] [talker]: Publishing: 'Talker says 3'
[INFO] [1746225416.011346700] [listener]: Listener heard: [Talker says 3]
[INFO] [1746225416.011466400] [listener]: Liveliness changed event:
[INFO] [1746225416.012171400] [listener]:   alive_count: 0
[INFO] [1746225416.012558100] [listener]:   not_alive_count: 0
[INFO] [1746225416.012924200] [listener]:   alive_count_change: -1
[INFO] [1746225416.013306300] [listener]:   not_alive_count_change: 0
```

Check `deadline`

```
C:\pixi_ws>ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
[INFO] [1746225523.388043600] [talker]: Talker starting up
[INFO] [1746225523.392050800] [listener]: Listener starting up
[INFO] [1746225523.900237400] [talker]: Publishing: 'Talker says 0'
[INFO] [1746225523.900714500] [listener]: Listener heard: [Talker says 0]
[INFO] [1746225524.416264300] [talker]: Publishing: 'Talker says 1'
[INFO] [1746225524.418626100] [listener]: Listener heard: [Talker says 1]
[INFO] [1746225524.900586700] [talker]: Publishing: 'Talker says 2'
[INFO] [1746225524.905199100] [listener]: Listener heard: [Talker says 2]
[INFO] [1746225525.400412700] [talker]: Publishing: 'Talker says 3'
[INFO] [1746225525.402695400] [listener]: Listener heard: [Talker says 3]
[INFO] [1746225525.900105900] [talker]: Publishing: 'Talker says 4'
[INFO] [1746225525.902719800] [listener]: Listener heard: [Talker says 4]
[INFO] [1746225526.400242800] [talker]: Publishing: 'Talker says 5'
[INFO] [1746225526.402340900] [listener]: Listener heard: [Talker says 5]
[... 12 lines cut ...]
[INFO] [1746225530.228353200] [talker]: Offered deadline missed - total 3 delta 1
[INFO] [1746225530.231006200] [listener]: Requested deadline missed - total 3 delta 1
[INFO] [1746225530.415786200] [talker]: Publishing: 'Talker says 10'
[INFO] [1746225530.422674000] [listener]: Listener heard: [Talker says 10]
[INFO] [1746225530.931666100] [talker]: Publishing: 'Talker says 11'
[INFO] [1746225530.933325000] [listener]: Listener heard: [Talker says 11]
[INFO] [1746225531.431359900] [talker]: Publishing: 'Talker says 12'
[INFO] [1746225531.433568400] [listener]: Listener heard: [Talker says 12]
[INFO] [1746225531.931632200] [talker]: Publishing: 'Talker says 13'
[INFO] [1746225531.933991100] [listener]: Listener heard: [Talker says 13]
[INFO] [1746225532.431342000] [talker]: Publishing: 'Talker says 14'
[INFO] [1746225532.434143900] [listener]: Listener heard: [Talker says 14]
[INFO] [1746225532.947127200] [talker]: Publishing: 'Talker says 15'
[INFO] [1746225532.949979500] [listener]: Listener heard: [Talker says 15]
[INFO] [1746225533.571962800] [talker]: Offered deadline missed - total 4 delta 1
[INFO] [1746225533.574094100] [listener]: Requested deadline missed - total 4 delta 1
[INFO] [1746225534.181710000] [talker]: Offered deadline missed - total 5 delta 1
[INFO] [1746225534.183464100] [listener]: Requested deadline missed - total 5 delta 1
[INFO] [1746225534.791105800] [talker]: Offered deadline missed - total 6 delta 1
[INFO] [1746225534.792049100] [listener]: Requested deadline missed - total 6 delta 1
[INFO] [1746225535.400284000] [talker]: Offered deadline missed - total 7 delta 1
[INFO] [1746225535.402725000] [listener]: Requested deadline missed - total 7 delta 1
[INFO] [1746225535.403524900] [talker]: Publishing: 'Talker says 16'
[INFO] [1746225535.404321000] [listener]: Listener heard: [Talker says 16]
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde