ros2/kilted_tutorial_party #232: Executables in `quality_of_service_demo_cpp`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-03:
- Check lifespan
```
legends@4legends-ASUS:~$ ros2 run quality_of_service_demo_cpp lifespan
Usage for lifespan:
lifespan lifespan_duration [--history history_depth] [--publish-count publish_count] [--subscribe-after subscribe_after] [-h]
required arguments:
lifespan duration: Duration in positive integer milliseconds of the Lifespan QoS setting.
optional arguments:
-h : Print this help message.
--history history : The depth of the Publisher's history queue - the maximum number of messages it will store for late-joining subscriptions. Defaults to 10
--publish-count publish_n_messages : How many messages to publish before stopping. Defaults to 5
--subscribe-after subscribe_after_duration : The Subscriber will be created this long in positive integer milliseconds after application startup. Defaults to 5000

``` 
- Check liveliness
```
legends@4legends-ASUS:~$ ros2 run quality_of_service_demo_cpp liveliness
Usage for liveliness:
liveliness lease_duration [--policy liveliness_policy] [--topic-assert-period topic_assert_liveliness_period] [-h]
required arguments:
lease_duration: Duration in positive integer milliseconds after which an inactive Publisher is considered not-alive. 0 means never.
optional arguments:
-h : Print this help message.
--policy liveliness_policy : You may specify AUTOMATIC, or MANUAL_BY_TOPIC. Defaults to AUTOMATIC
--topic-assert-period topic_assert_period : How often the Publisher will assert the liveliness of its Publisher , in positive integer milliseconds. Defaults to 0 (never)
--kill-publisher-after kill_publisher_after : Kill the publisher after this amount of time (in uint milliseconds). In AUTOMATIC - destroy the whole node. In MANUAL_BY_TOPIC, stop topic liveliness assertion. Defaults to 3000

``` 
- Check deadline
```
legends@4legends-ASUS:~$ ros2 run quality_of_service_demo_cpp deadline
Usage for deadline:
deadline deadline_duration [--publish-for publish_for] [--pause-for pause_for] [-h]
required arguments:
deadline_duration: Duration in positive integer milliseconds of the Deadline QoS setting.
optional arguments:
-h : Print this help message.
--publish-for publish_for : Duration to publish (in positive integer milliseconds) until pausing the talker. Defaults to 5000.
--pause-for pause_for : Duration to pause (in positive integer milliseconds) before starting to publish again. Defaults to 1000.
```

fujitatomoya, 2025-05-13:
@4-legends thanks for trying this. actually you are just printing the help message as we can see, the required arguments should be given. sorry, the issue header does not say anything like this...

```console
root@tomoyafujita:~# ros2 run quality_of_service_demo_cpp lifespan 100
[INFO] [1747113830.654721388] [talker]: Talker starting up
[INFO] [1747113830.655133417] [listener]: Listener starting up
[INFO] [1747113831.155374886] [talker]: Publishing: 'Talker says 0'
[INFO] [1747113831.655386208] [talker]: Publishing: 'Talker says 1'
[INFO] [1747113832.155398695] [talker]: Publishing: 'Talker says 2'
[INFO] [1747113832.655393051] [talker]: Publishing: 'Talker says 3'
[INFO] [1747113833.155410409] [talker]: Publishing: 'Talker says 4'
^C[INFO] [1747113842.252355907] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747113842.252630561] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~# ros2 run quality_of_service_demo_cpp liveliness 100
[INFO] [1747113861.775983636] [talker]: Talker starting up
[INFO] [1747113861.776476726] [listener]: Listener starting up
[INFO] [1747113862.276731972] [talker]: Publishing: 'Talker says 0'
[INFO] [1747113862.277205558] [listener]: Listener heard: [Talker says 0]
[INFO] [1747113862.277299598] [listener]: Liveliness changed event:
[INFO] [1747113862.277321145] [listener]:   alive_count: 1
[INFO] [1747113862.277337904] [listener]:   not_alive_count: 0
[INFO] [1747113862.277355039] [listener]:   alive_count_change: 1
[INFO] [1747113862.277369472] [listener]:   not_alive_count_change: 0
[INFO] [1747113862.376414346] [listener]: Liveliness changed event:
[INFO] [1747113862.376523203] [listener]:   alive_count: 0
[INFO] [1747113862.376541646] [listener]:   not_alive_count: 1
[INFO] [1747113862.376556984] [listener]:   alive_count_change: -1
[INFO] [1747113862.376573655] [listener]:   not_alive_count_change: 1
[INFO] [1747113862.776734425] [talker]: Publishing: 'Talker says 1'
[INFO] [1747113862.777020609] [listener]: Listener heard: [Talker says 1]
[INFO] [1747113862.777073856] [listener]: Liveliness changed event:
[INFO] [1747113862.777110047] [listener]:   alive_count: 1
[INFO] [1747113862.777164504] [listener]:   not_alive_count: 0
[INFO] [1747113862.777188026] [listener]:   alive_count_change: 1
[INFO] [1747113862.777211972] [listener]:   not_alive_count_change: -1
[INFO] [1747113862.876376592] [listener]: Liveliness changed event:
[INFO] [1747113862.876501791] [listener]:   alive_count: 0
[INFO] [1747113862.876521504] [listener]:   not_alive_count: 1
[INFO] [1747113862.876537131] [listener]:   alive_count_change: -1
[INFO] [1747113862.876554582] [listener]:   not_alive_count_change: 1
[INFO] [1747113863.276749818] [talker]: Publishing: 'Talker says 2'
[INFO] [1747113863.277048146] [listener]: Listener heard: [Talker says 2]
[INFO] [1747113863.277144094] [listener]: Liveliness changed event:
[INFO] [1747113863.277176795] [listener]:   alive_count: 1
[INFO] [1747113863.277201769] [listener]:   not_alive_count: 0
[INFO] [1747113863.277252309] [listener]:   alive_count_change: 1
[INFO] [1747113863.277268608] [listener]:   not_alive_count_change: -1
[INFO] [1747113863.376384958] [listener]: Liveliness changed event:
[INFO] [1747113863.376503193] [listener]:   alive_count: 0
[INFO] [1747113863.376523625] [listener]:   not_alive_count: 1
[INFO] [1747113863.376539625] [listener]:   alive_count_change: -1
[INFO] [1747113863.376556023] [listener]:   not_alive_count_change: 1
[INFO] [1747113863.776782727] [talker]: Publishing: 'Talker says 3'
[INFO] [1747113863.777086796] [listener]: Listener heard: [Talker says 3]
[INFO] [1747113863.777182290] [listener]: Liveliness changed event:
[INFO] [1747113863.777230153] [listener]:   alive_count: 1
[INFO] [1747113863.777251003] [listener]:   not_alive_count: 0
[INFO] [1747113863.777270546] [listener]:   alive_count_change: 1
[INFO] [1747113863.777288804] [listener]:   not_alive_count_change: -1
[INFO] [1747113863.876414452] [listener]: Liveliness changed event:
[INFO] [1747113863.876539142] [listener]:   alive_count: 0
[INFO] [1747113863.876558752] [listener]:   not_alive_count: 1
[INFO] [1747113863.876574647] [listener]:   alive_count_change: -1
[INFO] [1747113863.876591526] [listener]:   not_alive_count_change: 1
^C[INFO] [1747113863.911390033] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747113863.911656572] [rclcpp]: signal_handler(signum=2)

[... 9 lines cut ...]
[INFO] [1747113882.715203838] [talker]: Offered deadline missed - total 3 delta 1
[INFO] [1747113882.715335192] [listener]: Requested deadline missed - total 3 delta 1
[INFO] [1747113882.814394311] [talker]: Offered deadline missed - total 4 delta 1
[INFO] [1747113882.814525336] [listener]: Requested deadline missed - total 4 delta 1
[INFO] [1747113882.913451411] [talker]: Offered deadline missed - total 5 delta 1
[INFO] [1747113882.913570712] [listener]: Requested deadline missed - total 5 delta 1
[INFO] [1747113882.917247406] [talker]: Publishing: 'Talker says 1'
[INFO] [1747113882.917526653] [listener]: Listener heard: [Talker says 1]
[INFO] [1747113883.016871403] [talker]: Offered deadline missed - total 6 delta 1
[INFO] [1747113883.017003426] [listener]: Requested deadline missed - total 6 delta 1
[INFO] [1747113883.116085431] [talker]: Offered deadline missed - total 7 delta 1
[INFO] [1747113883.116218333] [listener]: Requested deadline missed - total 7 delta 1
[... cut ...]
[INFO] [1747113883.314574114] [listener]: Requested deadline missed - total 9 delta 1
[INFO] [1747113883.413643147] [talker]: Offered deadline missed - total 10 delta 1
[INFO] [1747113883.413774892] [listener]: Requested deadline missed - total 10 delta 1
[INFO] [1747113883.417257235] [talker]: Publishing: 'Talker says 2'
[INFO] [1747113883.417540267] [listener]: Listener heard: [Talker says 2]
^C[INFO] [1747113883.422117489] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747113883.422317597] [rclcpp]: signal_handler(signum=2)

```

closed 2025-05-13 by fujitatomoya