ros2/kilted_tutorial_party #234: Executables in `quality_of_service_demo_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed checks for Issue #234 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

Note: Running the executables without arguments printed help/error messages as arguments are required. Testing was performed by adding example arguments (e.g., `lifespan 1000`, `liveliness 1000`, `deadline 600`).

- Ran `ros2 run quality_of_service_demo_py lifespan 1000`. Node started successfully and ran without errors, demonstrating lifespan behavior as expected.
- Ran `ros2 run quality_of_service_demo_py liveliness 1000`. Node started successfully and ran without errors, demonstrating liveliness event callbacks as expected.
- Ran `ros2 run quality_of_service_demo_py deadline 600`. Node started successfully and ran without errors, correctly demonstrating deadline missed events as expected.

All Python QoS demo executables appear functional when run with appropriate arguments. The issue description might need updating to include example arguments in the "Checks" section.

closed 2025-05-09 by mjcarroll