ros2/kilted_tutorial_party #237: Executables in `quality_of_service_demo_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-03:
I am interested.

Tenveis, 2025-05-03:
Following are the results:

- [ ]  Check `lifespan`
- [ ]  Check `liveliness`
- [ ]  Check `deadline`

After running the command as it is, i am getting the below output.
[image]

Following are the output after putting argument.

- [X]  Check `lifespan`
argument: 1
[image]

- [X]  Check `liveliness`
argument: 1
[image]

argument: 10
[image]

argument: 100
[image]

- [X]  Check `deadline`
argument: 100
[image]

closed 2025-05-05 by ahcorde