ros2/kilted_tutorial_party #239: Executables in `rosbag2`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `ros2 bag record -a`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag record -a
       ```
  
    
  </details>
- [ ] **Check `ros2 bag record <topic1> <topic2> <topic3>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag record <topic1> <topic2> <topic3>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag info <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info <bag_file>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag play <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag play <bag_file>
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-05:
I will grab this one too!

Tested with turtlesim_node.

Check ros2 bag record -a

```
[INFO] [1746471450.976195500] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1746471451.049794600] [rosbag2_recorder]: Listening for topics...
[INFO] [1746471451.089678900] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1746471451.116569200] [rosbag2_recorder]: Subscribed to topic '/turtle1/rotate_absolute/_action/status'
[INFO] [1746471451.156585200] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1746471451.187752500] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1746471451.216669300] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1746471451.249796100] [rosbag2_recorder]: Subscribed to topic '/turtle1/color_sensor'
[INFO] [1746471451.272224000] [rosbag2_recorder]: Subscribed to topic '/turtle1/pose'
[INFO] [1746471451.299213800] [rosbag2_recorder]: Subscribed to topic '/turtle1/rotate_absolute/_action/feedback'
[INFO] [1746471451.330592800] [rosbag2_recorder]: Recording...
[INFO] [1746471457.882259600] [rosbag2_recorder]: Pausing recording.
[INFO] [1746471459.314259500] [rosbag2_recorder]: Resuming recording.
[INFO] [1746471460.614271400] [rosbag2_recorder]: Pausing recording.
```

Check ros2 bag record /turtle1/color_sensor /turtle1/pose

```
[WARN] [ros2bag]: Positional "topics" argument deprecated. Please use optional "--topics" argument instead.
[INFO] [1746471499.065691700] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1746471499.134326900] [rosbag2_recorder]: Listening for topics...
[INFO] [1746471516.252163500] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1746471516.270930700] [rosbag2_recorder]: Subscribed to topic '/turtle1/color_sensor'
[INFO] [1746471516.289682000] [rosbag2_recorder]: Subscribed to topic '/turtle1/pose'
[INFO] [1746471516.292421000] [rosbag2_recorder]: Recording...
[INFO] [1746471516.302328700] [rosbag2_recorder]: Pausing recording.
[INFO] [1746471516.334050400] [rosbag2_recorder]: All requested topics are subscribed. Stopping discovery...
[INFO] [1746471517.821371100] [rosbag2_recorder]: Resuming recording.
```

Check ros2 bag info rosbag2_2025_05_05-15_57_21

```
Files:             rosbag2_2025_05_05-15_57_21_0.mcap
Bag size:          42.4 KiB
Storage id:        mcap
ROS Distro:        kilted
Duration:          2.805451200s
Start:             May  5 2025 15:57:37.791705200 (1746471457.791705200)
End:               May  5 2025 15:57:40.597156400 (1746471460.597156400)
Messages:          205
Topic information: Topic: /events/write_split | Type: rosbag2_interfaces/msg/WriteSplitEvent | Count: 0 | Serialization Format: cdr
                   Topic: /parameter_events | Type: rcl_interfaces/msg/ParameterEvent | Count: 0 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 11 | Serialization Format: cdr
                   Topic: /turtle1/color_sensor | Type: turtlesim_msgs/msg/Color | Count: 97 | Serialization Format: cdr
                   Topic: /turtle1/pose | Type: turtlesim_msgs/msg/Pose | Count: 97 | Serialization Format: cdr
Services:          0
Service information:
Actions:           1
Action information:
  Action: /turtle1/rotate_absolute | Type: turtlesim_msgs/action/RotateAbsolute | Topics: 2 | Service: 3 | Serialization Format: cdr
    Topic: feedback | Count: 0
    Topic: status | Count: 0
    Service: send_goal | Event Count: 0
    Service: cancel_goal | Event Count: 0
    Service: get_result | Event Count: 0
```

Check ros2 bag play rosbag2_2025_05_05-15_57_21

```
[INFO] [1746471631.277049900] [rosbag2_player]: Set rate to 1
[INFO] [1746471631.998678100] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1746471632.011443200] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1746471632.026856100] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1746471632.047047800] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1746471632.063338000] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
Progress bar enabled at 3 Hz.
Progress bar [?]: [R]unning, [P]aused, [B]urst, [D]elayed, [S]topped
[INFO] [1746471633.091078500] [rosbag2_player]: Playback until timestamp: -1
[INFO] [1746471637.134732700] [rosbag2_player]: Pausing play.

====== Playback Progress ======
[1746471457.791705132] Duration 0.00/2.81 [P]
```

Double checked that expected topics are present by running ros2 topic list in a second terminal.

All executables are operational. All set to close!

Yadunund, 2025-05-05:
Thanks for your contribution!

closed 2025-05-05 by Yadunund