ros2/kilted_tutorial_party #242: Executables in `rosbag2`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `ros2 bag record -a`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag record -a
       ```
  
    
  </details>
- [ ] **Check `ros2 bag record <topic1> <topic2> <topic3>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag record <topic1> <topic2> <topic3>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag info <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info <bag_file>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag play <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 bag play <bag_file>
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
- [x] ros2 bag record -a

[image]

- [x] ros2 bag record <topic1> <topic2> <topic3>

[image]

- [x] ros2 bag info <bag_file>

```
root@cluttered:~# ros2 bag info rosbag2_2025_05_03-00_35_21/

Files:             rosbag2_2025_05_03-00_35_21_0.mcap
Bag size:          21.2 KiB
Storage id:        mcap
ROS Distro:        kilted
Duration:          57.933898934s
Start:             May  3 2025 00:35:21.425681297 (1746212721.425681297)
End:               May  3 2025 00:36:19.359580231 (1746212779.359580231)
Messages:          124
Topic information: Topic: /number | Type: std_msgs/msg/Int32 | Count: 58 | Serialization Format: cdr
                   Topic: /number2 | Type: std_msgs/msg/Int32 | Count: 58 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 8 | Serialization Format: cdr
Services:          0
Service information: 
Actions:           0
Action information: 
root@cluttered:~# 
```

- [x] ros2 bag play <bag_file>

[image]

[image]

RMW output after shutdown.

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde