ros2/kilted_tutorial_party #244: Executables in `rqt`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-04:
Hello, when I launched rqt and added "plot", it complained about not finding a usable plot type.

```  
File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_plot\data_plot\__init__.py", line 153, in __init__
    raise RuntimeError(
RuntimeError: No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.4.0) or Python-Qwt5.

Traceback (most recent call last):
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_topic\topic_widget.py", line 123, in refresh_topics
    topic_list = self._node.get_topic_names_and_types()
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rclpy\node.py", line 2166, in get_topic_names_and_types
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
```

I installed matplotlib using `pip install matplotlib` and now it works fine

[image]

ahcorde, 2025-05-12:
Thank you for testing