ros2/kilted_tutorial_party #246: Executables in `rqt`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
`rqt` was working outside the pixi shell.

[image]

But in pixi shell the `rqt` hang when a plugin is loaded. 

[image]

suchetanrs, 2025-05-04:
Hello, when I launched rqt and added "plot", it complained about not finding a usable plot type.

```  
File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_plot\data_plot\__init__.py", line 153, in __init__
    raise RuntimeError(
RuntimeError: No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.4.0) or Python-Qwt5.

Traceback (most recent call last):
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_topic\topic_widget.py", line 123, in refresh_topics
    topic_list = self._node.get_topic_names_and_types()
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rclpy\node.py", line 2166, in get_topic_names_and_types
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
```

I installed matplotlib using `pip install matplotlib` and now it works fine

@pradyum was this your issue?

The output is below:

[image]

Aniketkumarroy, 2025-05-10:
I launched rqt from pixi shell and it is working fine

[image]
I loaded different plugins and it worked except for `plot` and `python console`

[image]

this is the terminal output when I tried to load a python console
```RosPluginProvider._parse_plugin_xml() plugin file "C:\pixi_ws\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\pixi_ws\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
QLayout::removeWidget: Cannot remove a null widget.
PluginHandlerDirect._restore_settings() plugin "rqt_py_console/PyConsole#1" raised an exception:
Traceback (most recent call last):
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\qt_gui\plugin_handler_direct.py", line 121, in _restore_settings
    self._plugin.restore_settings(plugin_settings_plugin, instance_settings_plugin)
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_py_console\py_console.py", line 83, in restore_settings
    self._switch_console_widget()
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_py_console\py_console.py", line 69, in _switch_console_widget
    self._console_widget = PyConsoleWidget(self._context)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_py_console\py_console_widget.py", line 58, in __init__
    self.py_console.exit.connect(context.close_plugin)
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PyConsoleTextEdit' object has no attribute 'exit'```