ros2/kilted_tutorial_party #247: Executables in `rqt`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

NickTziaros, 2025-05-02:
## Setup
Used the ros-testing:kilted-desktop image.
[image]

-  If run with 

`docker run --rm=true -ti ghcr.io/sloretz/ros-testing:kilted-desktop bash`

[image]

- If run with 

`docker run -it   --env="DISPLAY=$DISPLAY"   --env="QT_X11_NO_MITSHM=1"   --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw"   --gpus all   --privileged   --device /dev/dri   ghcr.io/sloretz/ros-testing:kilted-desktop`

[image]

✅ As expected

## The error is Docker related. Other than that the functionality works as expected.

fujitatomoya, 2025-05-13:
the error is that application cannot find the window system such as X or wayland. these errors are expected to be printed if docker namespace does not have appropriate socket to the window system running in the host system.

closed 2025-05-13 by fujitatomoya