ros2/kilted_tutorial_party #249: Executables in `rttest`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Run pendulum_demo to generate data**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo -f pendulum_demo_results
       ```
  
    
  </details>
- [ ] **Check `rttest_plot`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rttest rttest_plot -s pendulum_demo_results
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

leander-dsouza, 2025-05-03:
I get an `Operation not permitted` issue when generating the data. I have sudo access on the terminal, but the issue still persists.

Command output:

[image]

[image]

JatinVira, 2025-05-10:
Hey,

Tested this out myself with the new docker image.
I had tried the pendulum experiment owing to other issues and thought the warning might be because of less RAM,
but I saw the same warning debug statements as @leander-dsouza reported.

Debug Statements
```
root@cab79ce69f0a:/# ros2 run pendulum_control pendulum_demo -f pendulumn_demo_result
Writing results to file: pendulumn_demo_result
Couldn't set scheduling priority and policy: Operation not permitted
mlockall failed: Cannot allocate memory
Couldn't lock all cached virtual memory.
Pagefaults from reading pages not yet mapped into RAM will be recorded.
Initial major pagefaults: 0
Initial minor pagefaults: 4394
rttest statistics for pendulumn_demo_result:
  - Minor pagefaults: 3
  - Major pagefaults: 0
  Latency (time after deadline was missed):
    - Min: 54198 ns
    - Max: 117760 ns
    - Mean: 64520.555000 ns
    - Standard deviation: 6223.742694

```

`Plotted results`

[image]

[image]

[image]

[image]

JatinVira, 2025-05-10:
If this is the expected behavior @ahcorde 
we can go ahead and `close this issue`.

I'd love to be an assignee with @leander-dsouza 

Thanks

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde