ros2/kilted_tutorial_party #251: Executables in `rttest`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Run pendulum_demo to generate data**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo -f pendulum_demo_results
       ```
  
    
  </details>
- [ ] **Check `rttest_plot`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rttest rttest_plot -s pendulum_demo_results
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

RishikesavanRamesh, 2025-05-02:
I am interested

akky20, 2025-05-08:
Tested !! Able to generate plots.. 

Terminal
[image]

Plots

[image]

[image]

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde