ros2/kilted_tutorial_party #252: Executables in `rviz2`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rviz2`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rviz2 rviz2
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Testing Issue #252 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

**Result: FAILED**

1. Ran `ros2 run rviz2 rviz2`.
2. The RViz splash screen appeared briefly, and the following INFO messages were printed to the console:Testing Issue #252 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

**Result: FAILED**

1. Ran `ros2 run rviz2 rviz2`.
2. The RViz splash screen appeared briefly, and the following INFO messages were printed to the console:

```text
[INFO] [1746452408.488029300] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746452408.488265700] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746452410.256224900] [rviz2]: Stereo is NOT SUPPORTED
```
3. However, the process then immediately exited with failure code `3221226505`, indicating a crash (likely an access violation). The main RViz window did not fully load or become interactive.

**Conclusion:** RViz2 fails to launch successfully on this setup, crashing shortly after initialization.

Will try to check on this

JatinVira, 2025-05-10:
@ShahazadAbdulla 

I think they added fixes in the new binaries for `Rviz2`
can you retry this experiment?
It must have resolved now.

ShahazadAbdulla, 2025-05-10:
Hi @JatinVira, great news!

I re-tested Issue #252 after manually replacing the `ros2-windows` directory in my Pixi environment with the contents from the official **ROS 2 Kilted Beta 2 (2025-05-07) binaries**. The test was performed on Windows 11 / amd64 with `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`.

**RViz2 now launches successfully!**
- The splash screen appeared.
- The INFO messages regarding Stereo and OpenGL version were present.
- The main Rviz2 window opened and became interactive.
- The previous crash (exit code 3221226505) no longer occurs.

It appears the fixes in the Beta 2 binaries have resolved this issue. Thanks for the heads-up!

[image]

JatinVira, 2025-05-11:
Yay,
Great job @ShahazadAbdulla

ahcorde, 2025-05-12:
Thank you for testing!

closed 2025-05-12 by ahcorde