ros2/kilted_tutorial_party #255: Executables in `rviz2`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `rviz2`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rviz2 rviz2
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-09:
Pulled the latest docker image and now `Rviz2` is working

Here's a snap of the same

[image]

JatinVira, 2025-05-09:
My setup:

```
   PLATFORM INFORMATION
system           : Linux
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde