ros2/kilted_tutorial_party #256: Executables in `rviz2`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `rviz2`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rviz2 rviz2
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I'm testing this.

zamantousif, 2025-05-09:
I tried to test this from inside the docker container with rmw zenoh and it worked fine.

[image]

ahcorde, 2025-05-09:
Thank you for tessting

closed 2025-05-09 by ahcorde