ros2/kilted_tutorial_party #257: Executables in `tf2_ros`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Testing Issue #256 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

**Results:**

- Check `tf2_echo`: Ran `ros2 run tf2_ros tf2_echo`. Correctly printed usage message and exited as arguments were missing. **(SUCCESS)**
- Check `tf2_monitor`: Ran `ros2 run tf2_ros tf2_monitor`. Started successfully, printed results header, and waited for TF data without errors. **(SUCCESS)**
- Check `static_transform_publisher`: Ran `ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map`. Started successfully, printed transform info, and ran until stopped. **(SUCCESS)**

**Interaction Test:**
- Ran `static_transform_publisher --frame-id world --child-frame-id map` in Terminal 1.
- Ran `ros2 run tf2_ros tf2_echo world map` in Terminal 2.
- **Result: FAILED.** `tf2_echo` continuously printed `Waiting for transform world -> map: Invalid frame ID "world" passed... frame does not exist` and defaulted to the identity transform. It did not receive the transform published by the static publisher.

**Conclusion:** While the individual `tf2_ros` executables start correctly, there appears to be an issue with TF discovery or communication when using `static_transform_publisher` and `tf2_echo` together with CycloneDDS on this Windows setup.

**Terminal Output for Failed Interaction Test (`tf2_echo`):**
<details>
<summary>Click to expand</summary>
 ```text
(pixi_ros2_rolling) C:\pixi_ws>ros2 run tf2_ros tf2_echo world map
[INFO] [1746449121.021695300] [tf2_echo]: Waiting for transform world -> map: Invalid frame ID "world" passed to canTransform argument target_frame - frame does not exist
At time 0.0

Translation: [0.000, 0.000, 0.000]

Rotation: in Quaternion (xyzw) [0.000, 0.000, 0.000, 1.000]
... (rest of identity transform output) ...
[INFO] [1746449124.623492000] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal: Interrupt```

</details>

Aniketkumarroy, 2025-05-13:
Hi I tested this on Windows 11, here are the terminal traces
### ***Setup***

[image]
### ***Running***
when ran `tf2_echo` without arguments

[image]
when ran `tf2_echo` with arguments

[image]

- [x] Check `tf2_echo`
- [x] Check `tf2_monitor`
- [x] Check `static_transform_publisher`

@ahcorde if everything's fine we can close this now

ahcorde, 2025-05-15:
Thank you for testing

closed 2025-05-15 by ahcorde