ros2/kilted_tutorial_party #259: Executables in `tf2_ros`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**Terminal-1:**

`
C:\pixi_ws>ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-13-16-47-421719-DESKTOP-8BMD1U3-10224' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-13-16-47-421719-DESKTOP-8BMD1U3-10224
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server.EXE-1]: process started with pid [9684]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [3888]
[INFO] [dummy_joint_states.EXE-3]: process started with pid [4776]
[INFO] [dummy_laser.EXE-4]: process started with pid [3968]
[dummy_laser.EXE-4] [INFO] [1746217011.321815500] [dummy_laser]: angle inc:     0.004363
[dummy_laser.EXE-4] [INFO] [1746217011.321886000] [dummy_laser]: scan size:     1081
[dummy_laser.EXE-4] [INFO] [1746217011.321891900] [dummy_laser]: scan time increment:   0.000000
[robot_state_publisher.EXE-2] [INFO] [1746217012.961528200] [robot_state_publisher]: Robot initialized
`

**Terminal-2:**

Check `tf2_echo`

`
C:\pixi_ws>ros2 run tf2_ros tf2_echo single_rrbot_link1 single_rrbot_link2
[INFO] [1746217101.394451700] [tf2_echo]: Waiting for transform single_rrbot_link1 ->  single_rrbot_link2: Invalid frame ID "single_rrbot_link1" passed to canTransform argument target_frame - frame does not exist
At time 1746217102.337482000
- Translation: [0.000, 0.100, 1.950]
- Rotation: in Quaternion (xyzw) [0.000, 0.207, 0.000, 0.978]
- Rotation: in RPY (radian) [0.000, 0.418, 0.000]
- Rotation: in RPY (degree) [0.000, 23.934, 0.000]
- Matrix:
  0.914  0.000  0.406  0.000
  0.000  1.000  0.000  0.100
 -0.406  0.000  0.914  1.950
  0.000  0.000  0.000  1.000
At time 1746217103.335514300
- Translation: [0.000, 0.100, 1.950]
- Rotation: in Quaternion (xyzw) [0.000, 0.475, 0.000, 0.880]
- Rotation: in RPY (radian) [0.000, 0.990, 0.000]
- Rotation: in RPY (degree) [0.000, 56.708, 0.000]
- Matrix:
  0.549  0.000  0.836  0.000
  0.000  1.000  0.000  0.100
 -0.836  0.000  0.549  1.950
  0.000  0.000  0.000  1.000
At time 1746217104.335756300
- Translation: [0.000, 0.100, 1.950]
- Rotation: in Quaternion (xyzw) [0.000, 0.072, 0.000, 0.997]
- Rotation: in RPY (radian) [0.000, 0.144, 0.000]
- Rotation: in RPY (degree) [0.000, 8.238, 0.000]
- Matrix:
  0.990  0.000  0.143  0.000
  0.000  1.000  0.000  0.100
 -0.143  0.000  0.990  1.950
  0.000  0.000  0.000  1.000
`

Check `tf2_monitor`

`
C:\pixi_ws>ros2 run tf2_ros tf2_monitor
Gathering data on all frames for 10 seconds...

RESULTS: for all Frames

Frames:
Frame: single_rrbot_camera_link, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_hokuyo_link, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_link1, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_link2, published by <no authority available>, Average Delay: 0.000404105, Max Delay: 0.00212979
Frame: single_rrbot_link3, published by <no authority available>, Average Delay: 0.000405187, Max Delay: 0.0021317

All Broadcasters:
Node: <no authority available> 14.3483 Hz, Average Delay: 0.0868077 Max Delay: 13.1337

RESULTS: for all Frames

Frames:
Frame: single_rrbot_camera_link, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_hokuyo_link, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_link1, published by <no authority available>, Average Delay: 13.1337, Max Delay: 13.1337
Frame: single_rrbot_link2, published by <no authority available>, Average Delay: 0.000351515, Max Delay: 0.00212979
Frame: single_rrbot_link3, published by <no authority available>, Average Delay: 0.000352382, Max Delay: 0.0021317

All Broadcasters:
Node: <no authority available> 14.2752 Hz, Average Delay: 0.0435537 Max Delay: 13.1337
`

Check `static_transform_publisher`

`
C:\pixi_ws>ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
[INFO] [1746217122.382243800] [static_transform_publisher_KF3lKUJgiU9Q8tFc]: Spinning until stopped - publishing transform
translation: ('0.000000', '0.000000', '0.000000')
rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
from 'world' to 'map'
`

**The checks have passed and `tf2_ros` executables are working fine.

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde