ros2/kilted_tutorial_party #261: Executables in `tf2_ros`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Works for me

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde