ros2/kilted_tutorial_party #262: Executables in `tlsf_cpp`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `tlsf_allocator_example`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tlsf_cpp tlsf_allocator_example
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-10:
Hey @ahcorde 
The testing instructions are a little unclear here.

Neverthless I gave it a shot to assist @suchetanrs 

Here's what I got
```
root@cab79ce69f0a:/# ros2 run tlsf_cpp tlsf_allocator_example
Intra-process pipeline is OFF.
root@cab79ce69f0a:/# ros2 run tlsf_cpp tlsf_allocator_example intra
Intra-process pipeline is ON.
root@cab79ce69f0a:/# 
```

My setup
```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-139-generic-x86_64-with-glibc2.39
release          : 5.15.0-139-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

```

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde